circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #1: Congratulations to the board(s)!

sandorkonya, 2022-06-14:
What is the current approximate bill of materials?

rajhlinux, 2022-07-24:
Yes would also like to congratulate such remarkable fully custom made camera.

It's impossible to buy bare bone Sony image sensors as a regular hobbyist consumer, can only buy Sony image sensors as a module.
Wonder where can one buy the buy bare bone Sony image sensors...

circuitvalley, 2022-07-26:
Currently i am preparing BOM with approximate price. So check back soon.

Yes bare sensors are hard to find i found my sensors from online retails specially Chinese websites. Or alternately you could get IMX477 sensors from  Raspberry Pi camera board

circuitvalley, 2022-10-09:
BOM cost is as follows with no tax and no shipping 
FX3 Board
QTY 1      QTY10      QTY 500
60             50              35
Crosslink NX
QTY 1      QTY10       QTY500
101           78             68
IMX290/462 Camera
QTY 1       QTY 10      QTY500
41             32              25

Total for Just PCBs
QTY 1           QTY10             QTY500
202 EUR       170 EUR          128 EUR

Enclosure and lens would be additional

closed 2022-10-09 by circuitvalley