circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #2: Hey, where can I buy your open source LIFCL-40-9BG256C chip?

lookembedded, 2022-08-17:
Hey, where can I buy your open source LIFCL-40-9BG256C chip? Can you share it with me? Or sell me one of your samples
It would be greatful if you could help me :)
Best regards,
Zhang Bo

circuitvalley, 2022-08-17:
Just few weeks ago they were in stock at mouser, Right now i think they are not in stock anywhere. If you really want to get started with LIFCL-40 there are devboards i think still available.
Regards

lookembedded, 2022-08-18:
Thanks for sharing the info. I don't want to go to the deveboard direction. I really like your 6-layers designed board style. Is that possible that you share one finished such camera with me. I would like to pay the cost.

circuitvalley, 2022-08-18:
If feel you can possibly drop an Email. I think i can possibly find a solution for you.

lookembedded, 2022-08-18:
Thanks. This is my email: jxh832027@gmail.com. Looking forward to your email.

circuitvalley, 2022-08-19:
Your Email appears to be incorrect.

lookembedded, 2022-08-19:
sorry，my bad. It is jxh831027@gmail.com

SinanAkkoyun, 2023-01-13:
Hi, I highly appreciate the work and nice design you have done!
Would it be ok for you if you could please also build me this camera? I will pay all costs and fees.

My mail is sinan@akkoyun.xyz

Thank you very much.
Best Regards,
Sinan

lookembedded, 2023-01-15:
Sorry, it's not mine. I just retweeted it. It's from a fan in Germany

| |
张博
|
|
***@***.***
|
---- Replied Message ----
| From | ***@***.***> |
| Date | 1/13/2023 09:59 |
| To | ***@***.***> |
| Cc | bo ***@***.***> ,
***@***.***> |
| Subject | Re: [circuitvalley/USB_C_Industrial_Camera_FPGA_USB3] Hey, where can I buy your open source LIFCL-40-9BG256C chip? (Issue #2) |

Hi, I highly appreciate the work and nice design you have done!
Would it be ok for you if you could please also build me this camera? I will pay all costs and fees.

My mail is ***@***.***

Thank you very much.
Best Regards,
Sinan