circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #9: Does anyone success build the FPGA source code?

INDNextHenry, 2023-05-15:
1. Lattice Radiant Software. 
[image]

2. Create the project contains the all source code.
3. try to build the prroject, it failed.
4. [image]

But It says a syntax error:  `Industrial_Camera_FPGA/source/impl_1/mipi_csi_rx_raw_depacker_8b2lane.v(134): syntax error near ). VERI-1137`.  

Any help thanks a lot! 
[Industrial_Camera_FPGA.zip](https://github.com/circuitvalley/USB_C_Industrial_Camera_FPGA_USB3/files/11475406/Industrial_Camera_FPGA.zip)

circuitvalley, 2023-05-16:
Try Creating a project without any tb_* files or disable tb_* file from project, You have wrong top module. There should be only one mipi_csi_16_nx.v as top.

INDNextHenry, 2023-05-17:
Hi Sir @circuitvalley, 

1. I removed the Testblocks source, only add /src/*, /src/Lattice_specfic/* to the project. 
2. set the `mipi_csi_16_nx.v` as top. 

I still got a error, not sure what going on, Any help is appreciated!!! 
[image]

INDNextHenry, 2023-05-18:
My bad, It has build success now. I got a typo issue.

closed 2023-05-18 by INDNextHenry

y-runwal, 2025-02-14:
@INDNextHenry , can you share hierarchy of successfully built project? I am facing some issues while managing files. Thank you.