circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #11: No ENABLE signal to boot the IMX290 sensor from FPGA

INDNextHenry, 2023-05-23:
Hi @circuitvalley, 
After program the fpga and usb3, finnaly I got a another issue, the IMX290 sensor board has no power. **The ENABLE signal was always low level**. I suspect that the fpga program is not normally started from the external flash, but I don't know how to verify this problem.
[image]
The Cypress FX3 USB3 serial port output log:
[image]
I think I should be very close to the final success, please help me, Thank you very much!!

INDNextHenry, 2023-05-23:
I think I loaded the `/Cypress_FX3_USB_Firmware/Image/USB_C_FX3_CV.img` was default suit for IMX477. So the IMX290 is not working.

circuitvalley, 2023-05-23:
You can compile img file for IMX290  your self or wait for a while I can provide you a .img file.

Regards

circuitvalley, 2023-05-24:
To fix compile issue, Include correct file in source. and include and exclude correct file in eclipse project as well. look on internet of including or excluding files from eclipse project

INDNextHenry, 2023-05-25:
Hi @circuitvalley 
I tried to build a fx3 project, then compile it.
[image]
It seems to work, then I got some *.img files.
[image]
Unfortunately I still don't get the ENABLE high signal of the IMX290 board.  The ENABLE signal of the IMX290 board is sent from the lattice FPGA. 
[image]
Although the FPGA programming is successful, I still need to investigate why the ENABLE signal is not enabled. 
[image]
I suspect that the FPGA is not functioning properly, the evidenced is the DONE signal not being pulled low as expected.
[image]
The LED2 quickly go off. 
[image]

INDNextHenry, 2023-05-25:
The latest status, I directly burn the program into the SRAM of the FPGA, the DONE signal is pulled low, and now the LED2 light is red.
[image]
However, the USB serial port gives a frame timer timeout warning, I guess because the frequency of the FX3 is set to 400MHZ? and the DMA needs to be set to 32K?
[image]

INDNextHenry, 2023-06-02:
In a words, I success programmed the fpga, but when power off the FPGA and restart, there is no EBABLE signal to IMX290.

nokiagx, 2023-06-03:
I'm not proficient in FPGA and I can't help you. I'm sorry！

closed 2023-09-18 by INDNextHenry