circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #13: Firmware for imx290 FX3

nzy20000618, 2023-07-07:
Hello, could you please send me the firmware of imx290 of fx3? I want to test the imx290 sensor I bought. Here is my email address: nn1275110583@gmail.com

rubenl1995, 2023-07-07:
Hi friend, I bought the sensor IMX290 but I can't power on camera. FX3 send frame by USB because I can watch by UART but with the application webcamoid not show nothing.

Thnak you!!

nzy20000618, 2023-07-07:
> 嗨，朋友，我买了传感器IMX290，但我无法打开相机电源。FX3 通过 USB 发送帧，因为我可以通过 UART 观看，但使用应用程序网络摄像头不会显示任何内容。
> 
> 谢谢你！！

Can your imx290 display images now? Can I add your contact information? You can contact me through my email: nn1275110583@gmail.com

circuitvalley, 2023-08-11:
There are few tips I could give you. How to debug if things are not working for you.Checking FPGA
1. Check fpga if it is outputting pixel clock , lsync and fsync. These signals are the most important if any of the pixel(data) lines are not correct then you will get a bad image but if any one of the Clock, Fsync or Lsync are not found you will get no image. This if any of these signals are not there then it may mean a bad solder joint. You can debug by debug signal from FPGA.
2. Check if FPGA outputs the same number of pixels on every frame. You would either need a Physical counter instrument  or a counter implemented in an FPGA. You need to count the exact number of pixels in a Frame. If FPGA output inconsistent number of byte every frame most UVC viewer will ignore frames, These may meen no image at all or massive drop in frame depending on how inconsistent your bytes are.

Checking FX3

1. FX3 part can be debugged by driving GPIO from GPIF or even C code. Debug frame sync Line sync etc. 

2. To check if you are receiving any data at all from the camera. Then use wireshark with USBPcap to capture USB packets. You can easily find these packets and also count how many bytes FPGA is sending. 

3. All the source files including sensor driver are present in repo. Only the sensor selected is included in the build. EZ USB suite allows you to select and deselect source files. EZ USB is eclipse based, Search on internet how to enable disable source in eclipse.4. If you are having a hard time resetting senor then it can be either permanently held out of reset with pull up or you can check how and which pins FPGA drive for resetting.
Sorry I do not have time for individual image builds. 
Regards