circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #14: Has anyone successfully reproduced this project?

INDNextHenry, 2023-08-07:
Has anyone successfully reproduced this project? I've been struggling with this project for a long time without success. 

The FPGA is programmed, and the fx3 runs successfully (the PC can recognize the usb device), but the image cannot be obtained using webcamod8.8. 

I guess there are two reasons: 1. There is a problem with the driver configuration of fx3 and imx219 or imx477. 2. There is a problem with the fpga configuration code.

Hope that someone who has successfully copied can provide suggestions or comments, thank you

circuitvalley, 2023-08-11:
There are few tips I could give you. How to debug if things are not working for you.Checking FPGA
1. Check fpga if it is outputting pixel clock , lsync and fsync. These signals are the most important if any of the pixel(data) lines are not correct then you will get a bad image but if any one of the Clock, Fsync or Lsync are not found you will get no image. This if any of these signals are not there then it may mean a bad solder joint. You can debug by debug signal from FPGA>2. Check if FPGA outputs the same number of pixels on every frame. You would either need a Physical counter instrument  or a counter implemented in an FPGA. You need to count the exact number of pixels in a Frame. If FPGA output inconsistent number of byte every frame most UVC viewer will ignore frames, These may meen no image at all or massive drop in frame depending on how inconsistent your bytes are.
Checking FX31. FX3 part can be debugged by driving GPIO from GPIF or even C code. Debug frame sync Line sync etc. 2. To check if you are receiving any data at all from the camera. Then use wireshark with USBPcap to capture USB packets. You can easily find these packets and also count how many bytes FPGA is sending. 3. All the source files including sensor driver are present in repo. Only the sensor selected is included in the build. EZ USB suite allows you to select and deselect source files. EZ USB is eclipse based, Search on internet how to enable disable source in eclipse.4. If you are having a hard time resetting senor then it can be either permanently held out of reset with pull up or you can check how and which pins FPGA drive for resetting.
Sorry I do not have time for individual image builds. 
Regards

INDNextHenry, 2023-08-14:
@circuitvalley  Thanks for the reply, that' helps. I will read every word carefully.

closed 2023-09-06 by circuitvalley