circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #15: Files for debayer simulation

MohamedWHassan, 2023-09-05:
Hello,
can you upload the files used in the debayer testbench such as
italy.bmp768x512.raw
italy.bmp768x512.raw.rgb
italy.bmp768x512.raw.yuv
italy.bmp768x512.raw.verify

circuitvalley, 2023-09-06:
Raw file is added rest of the files are auto generated during simulation. 
Commit. c76b455dc6708e8dc4d719f1ef93ee0b13e4a787

closed 2023-09-06 by circuitvalley

MohamedWHassan, 2023-10-17:
Hi,

Is it possible to tell me how did you manage to obtain the .RAW file?
I'm trying to find different RAW files with different bayer pattern but no luck so far.

Thanks

nokiagx, 2023-10-18:
Sorry, I don't know

kimstik, 2023-10-18:
> Hi,
> 
> Is it possible to tell me how did you manage to obtain the .RAW file? I'm trying to find different RAW files with different bayer pattern but no luck so far.
> 
> Thanks

https://swisscows.com/en/web?query=bayer+raw+image+sample
shows a lot of them:
https://stackoverflow.com/questions/27402422/where-can-i-download-raw-bayer-images
https://cs.brown.edu/courses/csci1290/labs/lab_raw/index.html

MohamedWHassan, 2023-10-18:
@kimstik 
Thank you for the reply. But it is hard to find these RAW bayer images that could be used in the testbench as I've been looking for days.

I found this [repo](https://github.com/zhengtianyu1996/bayer_image_process) which has a img2raw matlab script and with some modifications it worked nicely.

nokiagx, 2023-10-18:
kimstik：

I remember not posting any comments or suggestions in this field.
我记得我没有发表过这个领域的评论或建议。