circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #18: Anyone with extra hardware?

nategreco, 2023-10-22:
I'd like to use this project as a development/learning platform but I don't think it's practical for me to get a single piece of each board manufactured for myself - Does anyone have any extra hardware that they'd be willing to sell to someone in the US?

Thanks

INDNextHenry, 2023-10-23:
I have some, but it really hard to new fish to recopy this project. Beause it lacks bins && fw to verify the board.

nategreco, 2023-10-23:
Ugh, that's a good point. So you had boards made but have yet to verify they work?

INDNextHenry, 2023-10-23:
Already fight for months, little process made. I'm zero level on lattice fpga, not for reference, hahah.

nategreco, 2023-10-23:
I'm in a similar boat. I know a little verilog but it does seem daunting to spin up a lattice project from scratch.

Well if you had extras of all three boards and had any confidence they worked I'd be excited to buy them off you. I haven't looked into getting them fabricated myself but I figure both JLPCB and PCBway have minimum order quantities for assembled boards.

INDNextHenry, 2023-10-25:
> I'm in a similar boat. I know a little verilog but it does seem daunting to spin up a lattice project from scratch.
>
> Well if you had extras of all three boards and had any confidence they worked I'd [...]

I am happy to share my boards. I'm in China, where are your location?

nategreco, 2023-10-25:
US - Shipping will probably be expensive but if you're willing to ship it I'd cover the cost.

mca312, 2023-11-06:
Nate, did you the parts? I'm about to order some once I find the dimensions for the IMX273.

nategreco, 2023-11-06:
No, I haven't ordered anything yet. Are you buying yours turnkey assembled?

mca312, 2023-11-07:
I was going to order the image and FPGA board and place the parts myself.  But I'm looking at an IMX273 (or 252) and need to get the footprints for it first.  Also looks like there may be an issue in the design from the other open issue.

circuitvalley, 2023-12-02:
There will a second revision of this camera under development.

closed 2023-12-02 by circuitvalley