circuitvalley/USB_C_Industrial_Camera_FPGA_USB3 #19: The connector pins in the PCB board are mismatch

INDNextHenry, 2023-10-26:
Hi @circuitvalley 

I found that there are 3 pairs of pins on the upper connector of the FPGA and the lower connector of the USB3-FX3 that do not match, and the remaining 50 or so pins match normally. Is this a mistake or is it intentional design?

[image]

mca312, 2023-11-05:
Any luck on this? I’m about to start on a prototype

INDNextHenry, 2023-11-07:
> Any luck on this? I’m about to start on a prototype

No one mentioned this point, and very few people put this project into practical action.

circuitvalley, 2023-12-02:
Only names are in correct, This is fixed in constrain files pin assignment,

closed 2023-12-02 by circuitvalley